ISISComputingGroup/lewis #3: How will the Sim provide values to the Interface?

MikeHart85, 2016-04-06:
The Sim will be invoked by some sort of Interface (for example, a pcaspy server). Some states also involving timing/threading, so values may change delayed or independently from actions taken by the Interface.

How will the Interface get (and potentially set) values like current Speed, Phase, State, etc from the Sim?

Details to consider:

* Will the Sim provide getters and setters for all relevant values? 
* Should the Interface poll the getters periodically?
* Will there be a callback/observer mechanism to inform the Interface of changes the values? 
* A single call for "something changed" or individual/separate calls for each value?
* Will the new values be passed as a parameter, or are they to be retrieved via getter?

It seems currently we're leaning towards using an observer style pattern. This issue can also be used to flesh out ideas and research in that direction.

We should also decide whether we want getters and setters at all, or rely entirely on observation/callback mechanisms.

MichaelWedel, 2016-04-06:
For this my point of view is that we should offer both, meaning a callback/observer mechanism for Interfaces we have under our control, but also implement the values as getters/setters (actually preferably Python properties, I think that's considered good practice in Python).

That way other Interfaces may use the simulation in a different manner and maybe the callback/observer mechanism can even sit "on top" of the properties (e.g. use the setter function as callback or something like that) so implementing it wouldn't mean a lot of extra effort.

OwenArnold, 2016-04-08:
I think we have to consider the simulation as completely separate from the channel access protocol and other communication layers. At the same time, we want to keep that epics IOC as thin as possible. 

:+1: for properties

@MikeHart85 asked about timing in another issue, but I think the question is a good one. Do we use a polling/timing mechanism of some sort, or a call-back one? Who is responsible for setting frequency rate? How does this fit with the actual CHIC? We want to be able to replace that IOC or simulated device for a real one with minimal changes.
@MichaelWedel  maybe a sequence diagram to explain the designed sequence interaction between IOC and simulator would be useful?

MichaelWedel, 2016-04-08:
As I wrote earlier, I think it would be "pythonic" to offer more than one way to do it, so events can be emitted in addition to properties that can be queried by the sender. In the `implement_bearings`-branch I've started going in that direction (using [obsub](https://github.com/aepsil0n/obsub) for generating and subscribing to events for now), but there's nothing yet that reports some value in regular intervals. Good point about the update frequency setting, we'll have to think about where the responsibility for that should be and how to implement it best.

With regard to the IOC I don't see this particular point as critical because from the EPICS-client point of view it doesn't matter if the IOC gets its values by polling or via events, the client only cares about the PVs.

Once an IOC <-> CHIC protocol exists, we can implement that communication layer so it matches the behaviour of the CHIC. But again it should not matter how we get values etc. in the underlying layer.

MichaelWedel, 2016-04-18:
This is semi-solved now through #5 as well.

closed 2016-04-18 by MichaelWedel