ISISComputingGroup/lewis #4: Thread Safety Considerations

MikeHart85, 2016-04-06:
Since several states involve threading, we should consider what this means for thread safety.

For example: 

It looks like setting the speed will cause the state machine to periodically increase the speed in a separate thread to simulate "revving up". 

If the Interface is able to read the current speed while it is being written, this might lead to a synchronization issues. 

Another example might be getting the current state at exactly the same time as levitation finished and the state is transitioning.

MichaelWedel, 2016-04-06:
It's important that we make situations like the ones you describe thread safe so that we don't introduce unexpected behaviour into our simulation. This means we have to make sure to use mutexes etc. in the places that are affected by them.

I do however think that this mostly affects concrete implementations of certain behaviour, e.g. the default action to perform before/after a transition is nothing so for this case nothing needs to be done with regard to thread safety.

For the transitioning of states I'm not sure - does it matter if the state machine runs in the same thread as the client code?

MikeHart85, 2016-04-06:
This is where it will get tricky, I think.

Some parts of the state machine do look like they would be running in the client thread. But others won't.

Like in the example with the levitation finishing. If I'm reading the code correctly, the Sim currently simulates the time it takes to stabilize levitation by spawning a new thread that triggers a state transition after a specified time (say, 5 seconds). 

This all originally started by the client calling `Chopper.start()`. But, at this point, the client thread returns. A few seconds later, the transition is triggered by the spawned thread, independently from the client thread. It's unlikely, but if by chance the client thread tries to read the current state at exactly the same time, the result may be undefined.

Other places I see potential issues are: Speed setting, Phase locking, Stopping and Delevitating.

MichaelWedel, 2016-04-18:
Closing, because fixed through #5.

closed 2016-04-18 by MichaelWedel