ISISComputingGroup/lewis #5: Would it be better to implement the simulation without threading?

MikeHart85, 2016-04-08:
Since threading raises [some issues](https://github.com/DMSC-Instrument-Data/chopper-simulation/issues/4), I was wondering whether it might be better to implement the simulation without any threading at all.

Threads are currently used for implementing time delays, and would probably later also be used to implement gradually changing values and events that happen independently from / not triggered by client code.

Instead of using threads, we could make the simulation work based on a heartbeat / tick that gets called repeatedly by the client code. On each tick, the simulation would:

* Check how much time has passed
* Perform any delayed state transitions as needed
* Update any values that change periodically
* Invoke callbacks so the client can respond accordingly. 
* Et cetera?
* Return back to the client so it can do anything else it might want

Pcaspy works on a similar principle:

```
# Setup
server = SimpleServer()
server.createPV(prefix, pvdb)

# Processing loop
while True:
    server.process(0.1)
```

If we have a Simulation class that has a similar `process` method, we could just call both sequentially:

```
# Processing loop
while True:
    server.process(0.1)
    simulation.process(0.1)
```

This way we would avoid synchronization issues between client and simulation. There is only one thread, and the order things happen in is well defined and predictable.

It also provides a convenient way to control the update rate of values (I assume the hardware updates its outputs in regular intervals? Do we know what they are?).

This type of approach looks like it would fit well with pcaspy. Do we know if / how well it would work with the other interfaces we're planning on using?

MichaelWedel, 2016-04-08:
In the current throw-away prototype threads were used to avoid blocking while some internal thing is happening. That way clients can still query values while some aspect of the simulation is running in the background.

How would the ticks get propagated? Is there a kind of singleton that holds this state? Do all the classes we write need to know about this "event loop" explicitly? Who decides how often ticks occur?

I'm not sure how the hardware updates its values to the outside.

MikeHart85, 2016-04-08:
The way I had imagined it:

* The event loop is in the client code
* The client controls how often ticks occur by calling a `process` (or similar) method on a central `Simulation` class (or some other high level class like `Chopper`)
* This `process` method is then responsible for updating the simulation, triggering state transitions, etc
* Other parts of the simulation do not need to be aware of this (they're just being controlled by the `process` method)
* When the client, for example, sets the target speed, nothing actually happens at that point except a value being updated. Only when `process` is called does the simulation spring into action.

Regarding the central Simulation class, that's probably something to discuss in another issue... but I think we'll need something like that eventually to help set up and control simulation scenarios (for example, the chopper is faulty and overheats when it reaches X speed). That kind of code probably shouldn't be in the client, since we might want to use the same scenario with different clients.

MichaelWedel, 2016-04-08:
I'm not entirely sure I understand how for example the scenario in the current prototype would work with that system. Could you try to explain how the classes would look? Preferably in code, I think that would be easiest to understand.

MichaelWedel, 2016-04-18:
I'm closing this issue, because the way you solved it solved the issue.

closed 2016-04-18 by MichaelWedel