ISISComputingGroup/lewis #14: Package simulation prototype into Docker container

MichaelWedel, 2016-04-19:
We should have a Docker container like the motor container so that we can deploy multiple choppers.

MikeHart85, 2016-05-04:
#30 is semi-required for this. 

A flexible startup script will allow a different prefix to be passed through from the docker run command to the startup script inside the container. Otherwise, all instances will try to use the same PVs.

OwenArnold, 2016-05-10:
@MichaelWedel @MikeHart85. Some of my other colleagues at RAL/Diamond: David Michel, Nick Battam, Jon Holt, have formed a working group to figure out how to dockerise ioc's. I think there might be quite a lot of cross over with what we are trying to do, particularly when it comes to starting up many devicies/iocs. @MikeHart85 I've invited us to their working group.

MichaelWedel, 2016-05-13:
This could in principle be taken up now that the startup script has been sorted out.

MikeHart85, 2016-05-16:
`pcaspy` requires an EPICS base install, which will make the docker image quite beefy and time-consuming to build.

To cut down build time, we can create (or just use an existing) docker image with epics pre-installed as the base image for this one. This way, we just add plankton as a layer on top of it and just keep reusing the same EPICS base version layers.

Can't do much to get around it being quite large though.

MichaelWedel, 2016-05-16:
A docker image with EPICS pre-installed sounds like a good way to go. Maybe we can raise this on Tuesday with ICS to see whether they either already have something like that or would be interested in it in any way.

What does "quite large" mean in this context? Do you have an approximate size in MB?

MikeHart85, 2016-05-16:
As what user do we want to add this (and other) docker image(s)? 

Should we create a shared ESS account?

Note that the DockerHub currently doesn't have very good support for organizations as opposed to individual users. So if you want to publish our images as something like `ESS/plankton`, we'll need to create a user called ESS and share the password.

OwenArnold, 2016-05-16:
@MikeHart85 sounds fair enough. Creating a base image seems like the right way to handle these things.

MichaelWedel, 2016-05-16:
Would it be possible to use "DMSC-Instrument-Data" as user name? Then all our stuff would share this name somehow.

MikeHart85, 2016-05-16:
For size comparison:

- The [mikehart/motorsim](https://hub.docker.com/r/mikehart/motorsim/) image with the MCAG demo is 338 MB in size. 
- There's a [klauer/epics-docker](https://hub.docker.com/r/klauer/epics-docker/) image (from the NSLS-II by the looks of it) that's 563 MB. 
- And an [australiansynchrotron/epics](https://hub.docker.com/r/australiansynchrotron/epics/) that's only 168 MB.
- And I think this turned out to be one of ours? [sdmolloy/epics-basic](https://hub.docker.com/r/sdmolloy/epics-basic/) is 176 MB.

Not sure how the last two got it so (relatively) small, the Dockerfiles aren't available. Maybe just kept the binaries / deleted source and everything else non-essential?

MichaelWedel, 2016-05-16:
Interesting. But the EPICS source can't be that large or is it? Maybe it's okay to use the last one as a base? I could try and find him next time I'm in Lund and let him know that we're using his image.

MikeHart85, 2016-05-16:
I don't think dashes are supported in user names, as I've never seen that. And I don't think the names are case sensitive.

Should be able to make a "dmscinstrumentdata" user though. If we share the password for that with everyone else on the project, then everyone could push images to it.

Keep in mind, you need to write out the username anytime you want to pull, run, or push an image. Might get tedious with a long name, perhaps something shorter would be better. On the other hand, depending on platform, you may have auto-complete after the first pull.

MichaelWedel, 2016-05-16:
Hm, good point. "dmscid"?

MikeHart85, 2016-05-16:
A good chunk of those sizes is likely just the base image used. Most popular base images are ubuntu, centos and the like... they tend to be beefy from the start, and then you build on top of it.

It's also not just source, I imagine all of those images also built EPICS after downloading the source (and perhaps even got rid of the source after building?)

MikeHart85, 2016-05-16:
Also, do we care which version of EPICS exactly we're using?

MichaelWedel, 2016-05-16:
I don't think it matters as long as it works for `pcaspy`. That means, IIRC, no `3.15`.

PR #46 (2016-05-20): Docker Image Build Script
Closes #14.

This adds a Dockerfile (and associated .dockerignore) to allow building a docker image for Plankton.

To keep image size minimal, [Alpine Linux](https://hub.docker.com/_/alpine/) was used as the base image.

To optimize build speed and reusability, I've split it into three images:

- [dmscid/epics-base](https://hub.docker.com/r/dmscid/epics-base/): Just the Alpine base with EPICS base installed and minimal dependencies.
- [dmscid/epics-pcaspy](https://hub.docker.com/r/dmscid/epics-pcaspy/): Builds on top of `dmscid/epics-base` to add `python`, `pip` and ` [... cut]
closed 2016-05-20 by MichaelWedel