ISISComputingGroup/lewis #17: How should StateMachine splitting be implemented?

MikeHart85, 2016-04-19:
In reference to #13.

We should think about what the best way to implement split state machines / subcomponents would be. 

When writing the StateMachine class, I had two possible approaches in mind:

### Extension Approach

The first approach would be to have only one StateMachine instance per Device, and implement partial / subcomponent functionality by dynamically modifying that StateMachine to extend it.

This might be implemented using a Decorator Pattern or similar. E.g.:

```Python

class MagneticBearingComponent(Component):
  def __init__(self, csm, from_state, to_state):
    csm.remove_transition(from_state, to_state)
    csm.extend({... add some states between from_state and to_state ...})

  def _on_entry_levitate(self, dt):
    ...
  ...

class Chopper(Device):
  def __init__(self, ...):
    self._csm = StateMachine({... set up basic chopper state machine ...})
    self._magbear = MagneticBearingComponent(self._csm, 'init', 'idle')
```

This would require some way of telling the StateMachine that some events are implemented in a different object. Right now, it only supports one target object for raising events.

### Nesting Approach

The second approach would be for each subcomponent to have its own state machine (and essentially be a `Device` on its own). The subcomponent state machines would, conceptually, be nested within a state in the parent state machine. While the parent is in that state, it calls the `process()` of the subcomponent every cycle.

```Python

class MagneticBearings(Device):
  def __init__(self, ...):
    self._csm = StateMachine({...})
  ...

class Chopper(Device):
  def __init__(self, ...):
    self._csm = StateMachine({...})
    self._magbear = MagneticBearings()

  def _in_state_bearings(self, dt):
    self._magbear.process(dt)
```

Something to watch out for here: Depending on which direction we came from when we arrived in the `bearings` state, we may want to start in a different state in `MagneticBearings` (e.g.: `resting` or `levitating`).

### Other Approaches

These are the two I had in mind, but perhaps there's a different way of accomplishing this?

MichaelWedel, 2016-04-20:
I think the nesting approach is cleaner in the sense that it encapsulates the components from each other with less information. The "inner" statemachine does not need to know of the existance of the outer one at all in this approach.

MikeHart85, 2016-04-20:
Agreed, I like the nesting option better as well.

There are a few things to consider though:

1. As mentioned above, what state we want to start in within the inner statemachine may depend on how we got to the containing state in the parent statemachine. This isn't necessarily an issue. As long as we don't reset the inner statemachine, it will still be in the same state as when we left it. But that does create the requirement that we can only enter the nested statemachine the same way we left it.

2. The nested statemachine will only receive a `process()` call when the parent statemachine is in the containing state. For state processing itself this isn't an issue, because states shouldn't be processed anyway unless you're in them. But if the component wants to do some pre- or post-processing calculations, it may want to do those on every cycle... and as-is this will not happen. We'll either have to provide a separate way to trigger those, or design components with this limitation in mind.

3. The nesting model has strong encapsulation, which is generally a good thing. But how do we handle situations when the nested device has behaviour that depends on properties of the parent device? For example, perhaps the bearings should fail if `speed` exceeds a certain maximum. Perhaps MagneticBearings have a higher threshold than BallBearings.

MichaelWedel, 2016-04-20:
So to sum up at least one part: We both prefer the nesting approach, so I suggest we go with that.

Maybe we have our component model wrong. Maybe `Bearing` should be a part of `Drive` and not a sibling inside `Chopper`. For that specific case anyhow.

For the general case where state may be co-dependent, we need to find a good way of representing such a relationship. But since the state machine supports adding transitions, we can sort of already do that by adding a member function of sibling A to a transition of sibling B. The best way to do this is something we can surely find out later on.

MichaelWedel, 2016-04-28:
Do we still need to discuss this further?

MikeHart85, 2016-05-03:
Potentially... So far we haven't run into the limitations I mentioned, but I think we still might as we start simulating more devices and/or making the simulations more detailed.

Point 3 is now effectively resolved, since we've introduced device Contexts. Nested devices can just share the same context, or at least give each other access to theirs.

Points 1 and 2 are still things we need to watch out for.

But we might want to close this issue and, if needed, open another since we've gone quite far from the original topic with this. As far as the original question goes, we've decided to use the Nested approach and have already implemented it.

closed 2016-05-03 by MichaelWedel