ISISComputingGroup/lewis #18: CanProcessComposite should support pre- and post-processing

MikeHart85, 2016-04-19:
There will likely be situations where we want a Device to do something global before and/or after letting its state machine process a cycle.

For example, we may want to check for error conditions that are common to all/most states before processing a cycle. Or we may want to update some outputs after every cycle, regardless what state we are in.

Currently the `CanProcessComposite` class has a `compositeProcess()` method that is called before calling `process()` on all registered components. But there is nothing that is called after processing.

We could rename `compositeProcess()` to `beforeProcess()` and add a corresponding `afterProcess()`.

Afterthought: 
Will we ever have more than one thing to add to the `CanProcessComposite`? If we use the nesting approach described in #17, we'll have addition subcomponents with a `process()` method, but we will only want to call that in certain specific states and not in the general Device `process()`. If we use the extension approach, I think we'll only ever have one StateMachine instance to invoke. Is it worth outsourcing the process logic for that at all?

MichaelWedel, 2016-04-20:
I agree on the post- and pre-processing, those could go into `CanProcess.process` as "optional methods" like `compositeProcess()` is at the moment. The actual processing could then be done in `doProcess` and `CanProcessComposite` would implement that the way it implements `process` now.

I'm not sure there will always ever only be one thing we want to call `process` on, even within one state of a state machine.

Imagine we have a `ChopperDrive` with a state machine that has a `phase_locked` state. In that state we model the current phase of the chopper, for example by returning fluctuating the phase by some random amount each time the function is called. But at the same time we need to model something unrelated, let's say the temperature in the chopper housing (which for the sake of argument also fluctuates on each cycle).

The state handler in that state would need to call both functions. Is that covered at the moment?

To achieve this we could introduce a `__call__`-method on `CanProcess` that calls `process`, then we can install these as `in_state`-handlers in the state machine.

MichaelWedel, 2016-04-20:
What I have in mind is something like this:

```
class Chopper(CanProcess, ...):
  def __init__(self, bearing, drive):
    self._bearing = bearing # Possibly with state machine inside, implements CanProcess
    self._drive = drive # Also with some process logic inside or further contained devices
    self._state_machine = StateMachine({
# transitsions..
      'in_state_handlers': {
        'bearings': self._bearing,
        'drive': self._drive,
      })

  def doProcess(self, dt):
    self._state_machine.process(dt)
```

MikeHart85, 2016-04-20:
Re: the `ChopperDrive` example, this is how I would have imagined it (assuming nested statemachines and without mixins to illustrate what happens where):

```Python
class ChopperDrive(...):
  def __init__(self):
    self._state_machine = StateMachine(self, {
      # Some setup that contains a 'phase_locked' state
    })
  ...
  ...
  def process(self, dt):
    self._state_machine.process(dt)

  def _in_state_phase_locked(self, dt):
    # Some code or a function call to fluctuate phase randomly
    # Some code or a function call to calculate temperature of chopper housing

class Chopper(...):
  def __init__(self):
    self._state_machine = StateMachine(self, {
      # Over-arching statemachine, including a 'bearings' and a 'drive' state or similar
    })
    self._drive = ChopperDrive(...) # or passed in as a parameter
    self._bearings = MagneticBearings(...) # or param
  ...
  ...
  def process(self, dt):
    self._state_machine.process(dt)

  def _in_state_drive(self, dt):
    self._drive.process(dt)

  def _in_state_bearings(self, dt):
    self._bearings.process(dt)
```

The client (or whatever manages the simulation) only calls `Chopper.process(delta)`. That then causes the `Chopper`s statemachine to look at what state it is in and, if appropriate, call one of the `_in_state_[drive|bearings]` handlers. 

Anything the `ChopperDriver` needs to do while it is in the `phase_locked` state just belongs in `ChopperDrive._in_state_phase_locked()`. There shouldn't be any kind of `process` call in there, unless there's another nested statemachine inside that state.

MichaelWedel, 2016-04-20:
That looks very similar, but somehow I think these `_in_state` handlers should belong to the state machine, because they are tightly connected to the state machine. What happens inside these handlers is however `Device`-logic (which may or may not involve another state machine, we don't know that).

So why not have the state machine generate `_in_state_...` handlers on itself instead of the object that is passed on to the constructor and instead offer to assign callable objects with `dt` parameter there? I think that might improve the separation.

MikeHart85, 2016-04-20:
As per Slack discussion, we'll add a way to override default handlers in the StateMachine initialization settings.

closed 2016-04-28 by MichaelWedel