ISISComputingGroup/lewis #19: StateMachine should support custom handlers and multiple handlers

MikeHart85, 2016-04-20:
As per discussion in #18.

The StateMachine class should allow the user to override default `on_entry`, `in_state` and `on_exit` handlers. Specifying more than one handler should also be possible.

This could be achieved by adding support for an initialization parameter such as the following:

```python
self._state_machine = StateMachine(self, {
  'initial': 'off',
  'states': {
    'off': [custom_on_entry, custom_in_state, custom_on_exit],
    'idle': {'in_state': self.only_custom_instate_others_default},
    'accelerating': {'in_state': [self.one_handler, something_else.another_handler]}
  },
  'transitions': {
    # any states not mentioned above get auto-created with default behaviour
  }
})
```

MichaelWedel, 2016-04-20:
One question though - does `StateMachine` really still need the reference to the owning object?

MikeHart85, 2016-04-20:
The `target` reference is currently used for two things:

* Current behaviour for state handlers is to try calling `target._[event_name]_[state_name](dt)` (if such a method has optionally been defined there). 
* If a transition condition is specified by name, rather than function reference or lambda, it tries callling `target._check_[condition_name]()`.

My idea was for the `'states'` option to allow overriding the handler behaviour, but keeping the current behaviour as a default that is used when no custom handler is specified. 

This way, we won't have to spell out every state handler for every state in the initialization (and always keep it updated whenever we add or remove state handler implementations). It also establishes a standard naming convention for handlers. 

We would then only use the `'states'` option when we want to do something unusual, or as a shortcut to avoid having to define a handler that only calls another function (similar to how you can shortcut transition conditions with a lambda).

MichaelWedel, 2016-04-21:
I suppose there is absolutely no way of not having to specify state handlers at all (and updating/adding them), so I guess it's a matter of taste. Personally I would prefer extending the initialisation so that the state machine does not rely on me typing out the handler name correctly.

The naming convention concerns me a bit in the sense that it kind of "spills over" to client code. If the convention is changed, all client code needs to change - putting that into the initialisation with only state names the client code is not affected by any change in the state machine code. A similar argument can be made for the state names. Changing state names means changing the class interface.

Maybe the target could be made the second argument to `__init__` and optional (`None` as default). Then it's the clients choice whether or not it wants the state machine to know about its owner or not.

MikeHart85, 2016-04-21:
We should be careful about adding too much flexibility and too many options. Every new option means increased complexity, every time we increase flexibility we also decrease predictability. One of the big benefits of rolling our own StateMachine is that we don't have to support as many scenarios as possible... instead, we can use it to enforce strict conventions that help keep things simple.

As you say, there's no way around handling state events. For every event type (entry, in state, exit), for every state, if we want something in the device to change, we must have something like this, somewhere:

```python
def some_function_name(self, dt):
  # Do stuff
  # Some more stuff
  # And maybe some function calls
```

The demo chopper has 11 of those. The "real" thing will almost certainly have a lot more. But this is a given; these functions contain device behaviour, and so it's just a matter of device complexity and how much detail we want to capture in the simulation. However, if we explicitly bind all of them to events, we'll additionally wind up with a long list of them in `__init__` that we have to maintain... but what do we gain from that?

```python
def _in_state_process_bearings(self, dt):
  self._bearings.process(dt)
```
turns into
```python
# Somewhere in __init__
'process_bearings': {'in_state': self._bearings.process}
```
Which is nice because we avoided boilerplate code.

But that also means that if we had a situation like this:
```python
def _in_state_process_bearings(self, dt):
  # Something that must happen before
  self._bearings.process(dt)
  # Something that must happen after
```
it will have to become
```python
# Somewhere in __init__
'process_bearings': {'in_state': [self._do_pre_bearings, self._bearings.process, self._do_post_bearings]} 

# Further down in the class somewhere
def _do_pre_bearings(self, dt):
  # Something that must happen before

# You won't find what happens between pre and post here... you have to look in __init__

def _do_post_bearings(self, dt):
  # Something that must happen after
```

My feeling is that an occasional boilerplate handler that just calls `process` on another class is the lesser evil. Most handlers won't do that anyway... they'll contain a bunch of math to calculate speed, temperature, track timers, etc.

MichaelWedel, 2016-04-21:
There's no reason why there couldn't be a method
```
def handleBearingsState(self, dt):
  # some pre code
  self._bearings.process(dt)
  # some post code
```
The thing that makes me uncomfortable is the forced naming. The multiple handlers scenario I had in mind was having two devices that do stuff independently:

```
'process_bearings': {'in_state': [self._bearings, self._chopper_housing]}
```
Their logic shouldn't be part of `Chopper` if it doesn't need to be.

MichaelWedel, 2016-04-21:
That being said, we should not spend too much time on this detail now. As long as states can have custom event handlers everything we need to do for now can be covered. Whatever is strange after that can be changed later on. It's not like we have a huge code-base that we need to be careful about at the moment.

MikeHart85, 2016-04-21:
What is wrong with forcing a naming convention? It would just ensure all device implementations follow a well-defined structure, making it easier to pick up someone else's code / maintain old code.

Maybe it's a matter of perspective... I don't think of the Device classes as clients of the StateMachine class. So the convention isn't spilling over, it is shared.

The Device classes implement the device state functionality. Each device is really a state machine. The StateMachine class just contains the functionality common to all devices... it helps manage transitions, raise events, and provides a framework to help ensure all device implementations follow the same structure. 

Perhaps inheritance would be a better model than composition? `class Chopper(StateMachine, ...)`?

MichaelWedel, 2016-04-21:
You are right that currently the devices implement the device state functionality. That is however only true for the simulation part of what we are doing (which we are doing now). So yes, simulated devices are driven by a state machine. But not all devices are necessarily simulated. In the end I think we should be able to implement a `Chopper` that talks to the relevant EPICS PVs instead of the state machine. That's why I think it's best to keep these aspects (devices vs. simulation) as separate as we can.

PR #21 (2016-04-25): Re #19 Added custom handler support and updated demo sim
Closes #19.
closed 2016-04-25 by MichaelWedel