ISISComputingGroup/lewis #32: Use-cases for Plankton

OwenArnold, 2016-05-10:
Use-cases mentioned by IBEX group.

1. Simulate the Device not the IOC. The IOCs all exist at ISIS, but we could modify the start-up scripts to point to a port on which the simulator is running. Plankton is already well enough separated to do this. But we need to provide different `Adapters`. Freddie mentioned that having a EPICS style `*.proto` file for say a serial connection may be a way of configuring `Adapters` to do this. 
1. Use Simulators to preview a script-job. Use the simulators to fully preview (I'll call this a dry-run mode) what a given script is going to do when executed.
1. Modify the delta-t to run script-jobs in an integration-test mode.
1. Reverse-engineer existing devices such as the ISIS SKF choppers to build up a device/hardware suite.

@MichaelWedel @MikeHart85 I think some of this would be a good thing to try together on the 6th Jun at ISIS.

MichaelWedel, 2016-05-10:
1. I described this scenario in the documentation I started writing last week, but only in an abstract way without any technology suggestions. I'll take a look at the `proto` file format, maybe there's already a simple way to do this or something similar in Python.
2. Sounds good, probably with a different delta t as mentioned in 3 to have a "time lapse".
3. @MikeHart85 and I talked about this briefly already, for some types of tests the real time should be as close to 0 as possible.
4. This would probably have to be done in a similar fashion as we've been doing it for the CHIC, it would need support from the ISIS chopper group. Although the CHIC is currently quite close to the SKF chopper I think.

Maybe even no. 4 would be a good exercise for the workshop on June 6th?

OwenArnold, 2016-05-10:
@MichaelWedel. I've asked David Keymer to provide us with the c# control code or documentation for one of the mark-3 choppers at ISIS. Making a simulator for this at the first day of the workshop was exactly my intention for this. @MikeHart85 prior to the workshop we'll create a qfsm file and sanity check it with David Keymer.

MichaelWedel, 2016-08-30:
I think we can close this now, there's a ticket for modifying delta t and the other parts have been addressed by implementation of the Linkam Device and the stream adapter.

closed 2016-08-30 by MichaelWedel