ISISComputingGroup/lewis #35: Modify startup script to specify delta t

MichaelWedel, 2016-05-12:
The startup script for simulations needs to accept delta t as a parameter from the user so that simulations can be run at different speeds.

This is important later on when we want to run different scenarios as a kind of "system test suite".

MikeHart85, 2016-05-12:
Might be something for a different ticket, but Freddie also mentioned he'd like to be able to change the delta t at runtime. For example, via a special PV.

MichaelWedel, 2016-05-12:
Yes I think that's a different ticket. Currently this is the scope of the adapter, so I see a few options:

1. Modify `EpicsAdapter` to add a special PV.
2. Provide another adapter that implements this functionality, say `RuntimeControllableEpicsAdapter` for fans of long names.

The problem with both options is that they would tie this functionality to EPICS, while in fact this seems (at least to me) like a valid requirement for any simulation, regardless of the protocol.

Probably we can think of this as a second way of running simulations. There could be a simple server which is controlled by a simple command language over a protocol, for example ZeroMQ. This would be trivial to translate to some other protocol like EPICS. Coincidentally this sounds a lot like something that exists in the live-data-prototype.

MichaelWedel, 2016-05-12:
This is how it's done in the live data prototype. There is an interface `Controllable` which defines what things that should be controllable at runtime via a network protocol should implement:

https://github.com/DMSC-Instrument-Data/live-data-prototype/blob/master/controllable.py

Then there is a server which exposes the defined properties and/or trigger functions via zmq:

https://github.com/DMSC-Instrument-Data/live-data-prototype/blob/master/parameter_control_server.py

Somewhere on my machine I have a `Controllable -> EPICS`-adapter.

MikeHart85, 2016-05-12:
It is a feature we will likely want on all protocols, in some form, but I don't think it warrants creating an additional layer of separation, especially if it involves an additional server/client relation. 

There isn't that much functionality to add... and it seems like many things may vary from adapter to adapter.

For example: What exactly happens when you double the delta T? The simulation will run twice as fast of course... But what about the outputs? Should they get pumped out twice as fast as well? Or stay at once per (realtime) second like they currently are in the epics adapter? Will the answer be the same for all adapters? Will some adapters need a hard limit on how fast they pump out data?

MichaelWedel, 2016-05-12:
This is exactly why I think separating this from the rest of the simulation is valid, it has to go into the general interface of `Adapter` so that a server like this can expose these parameters and their limits etc. uniformly.

Whatever approach we select in the end, it is well beyond passing a delta t parameter to the startup script.

MikeHart85, 2016-05-12:
Yeah, we should leave this discussion for when we get to that point... once we start implementing other adapters, like the ones Freddie wanted to try, it will probably be much clearer where we can optimize/organize better.

PR #93 (2016-09-20): Remove run-logic from Adapter
This fixes #92, fixes #35, fixes #87.

It is now no longer the responsibility of the adapter to actually start and run the simulation, this has been moved into `SimulationEnvironment`, as discussed in the design document earlier.

`SimulationEnvironment` provides control over the processing time in each cycle and how that maps to simulated time via a factor. It also keeps track of the total processing time, total simulation time and number of processing cycles as well as simulation cycles. It is possible to pause and resume the simulation. The simulation can also  [... cut]
closed 2016-09-20 by MikeHart85