ISISComputingGroup/lewis #47: Naming Conventions

MikeHart85, 2016-05-20:
We've been somewhat frivolous with naming conventions for functions.

Some functions use `lowercase_with_underscores()` while others use `camelCase()`. 

We should standardize and update before the code base grows much bigger.

I think we settled on camelCase when we briefly talked about this before? Any objections / other ideas?

MichaelWedel, 2016-05-21:
I vote for `CamelCase` for classes, `dromedarCase` for methods and `lowercase_with_underscores` for free functions and local variables.

MikeHart85, 2016-05-21:
Sounds good to me.

`CAPS_AND_UNDERSCORES` to indicate variables intended to be used as constants?

MichaelWedel, 2016-05-21:
Do we have that anywhere?

MikeHart85, 2016-05-21:
I don't think we do yet. Just think we should settle on a convention for later.

The `epics` variable defined in bindings.py should arguably only be used for lookup and not change at runtime... might apply to that.

There's no straightforward way to enforce constness in Python, but some people like to use all caps to signify intent.

Personally, I think I'd prefer leaving it lowercase, and not to make assumptions that aren't enforced anyway.

MichaelWedel, 2016-05-21:
I agree to the last sentence. Enum-values could be made upper case to make them easier to spot.

MichaelWedel, 2016-05-24:
I think the actual action out of this should be adding a `contributing` document somewhere where this goes into a "Coding standards" section.

MikeHart85, 2016-05-24:
Should we use the Wiki feature on Git for this and other documentation?

MichaelWedel, 2016-05-24:
In my opinion the best thing is having all documentation in the source tree. Then there's one place for everything and easier to keep track of. Mantid is moving towards this approach and it's much easier to maintain, test etc.

Whenever there is time I'm looking further into #10, for now I think we should just have a markdown file.

MichaelWedel, 2016-10-16:
This has been somewhat superseded by reality, we got rid of the `dromedarCase`-methods, so now all functions are `underscore_case`. Perhaps this should also be in a section in the contribution guidelines.

mattclarke, 2020-11-03:
See PEP8

closed 2020-11-03 by mattclarke