ISISComputingGroup/lewis #51: Docker networking on Windows/OSX

MikeHart85, 2016-05-25:
On Windows and OS X, docker must be run in a virtual machine, usually using Docker Machine.

This causes problems for us, because it changes the way networking works.

On Linux, you get a `docker0` network for all of your docker containers. This network defaults to the `172.17.0.0/16` range. `172.17.0.1` is the host machine. Running containers are then assigned, in sequence, `172.17.0.2`, `172.17.0.3`, etc when they are executed. The broadcast address `172.17.255.255` can be used to broadcast to all running containers.

Ports do not need to be `EXPOSE`d or forwarded when interacting with this network, because each container behaves like just another device on that network.

This is all ideal for simulating a network of simulated devices or IOCs.

When using Docker Machine, this picture changes drastically:

The `docker0` network only exists inside the virtual machine. The host has no access to it.

Instead, from the perspective of the host, you only have access to the "external" IP of the virtual machine on the Host-Only network (`192.168.99.100` by default). The `docker0` network is hidden somewhere behind that.

You can use the docker `EXPOSE` keyword (Dockerfile) or `-p`ublish parameter (command-line) to bind a specific port of an image or container to a specific port on that "external" IP of the virtual machine.

You can also set up a Bridged-Network, in addition to the Host-Only. But all that achieves is allowing other machines on your (real) network to connect to your virtual machine. You still need to `EXPOSE` or `-p`ublish.

This all works fine if you're looking to run a service like `apache` or `mysql` in a container. All you want is to be able to access the service on a particular port on a particular IP.

But we want to talk EPICS CA on a network, which revolves around broadcasting to the network and seeing who replies. So this is no good.

Ideally, we need to find a way to expose or emulate the `docker0` network on Windows and OS X.

Or find an EPICsy workaround to allow the host to talk CA to containers in the virtual machine.

MichaelWedel, 2016-05-25:
Could the CA Gateway be a solution to this problem? It would be a solution very much centered on EPICS, but I suppose that's okay for now given that we don't have any other adapter.

MikeHart85, 2016-05-25:
Yeah, Owen and I were thinking about Gateway as well... perhaps that's the only way.

It would need to run on the docker virtual machine, not inside the containers, right? So that it can listen on 192.168.99.100 and broadcast to 172.17.0.0/16 and vice versa? Would there need to be a Gateway running on the host machine as well?

I was hoping there might be a cleaner solution, that doesn't involve using a different EPICS layout on Windows and OS X compared to Linux.

MichaelWedel, 2016-05-25:
Or different layout in general. For all other adapters this would not work.

MikeHart85, 2016-05-25:
Other adapters might be easier (though still not perfect)

For TCP connections, for example, the `EXPOSE` and `--publish` features of docker would work fine. Since we just need an IP and a port to connect to, and those features give us just that.

It's mainly EPICS that's going to be painful since we need to be able to UDP broadcast to a network and likewise receive responses.

MichaelWedel, 2016-05-25:
You're right, but still I agree that it would be better to have solution that works everywhere.

MikeHart85, 2016-05-25:
We've scheduled a meeting with Freddie next week to discuss the Gateway option with him (and perhaps he'll have other ideas as well).

MikeHart85, 2016-06-16:
Docker is working on better native support on Windows and OS X, which might fix this issue for us. However, that is still in beta.

In the mean time, using Gateway to help forward traffic looks like a promising approach.

To avoid installing anything (let alone EPICS+Gateway) inside the docker-machine VM, it would be best to package Gateway as a docker container that is configured to act as a link between the docker0 network and a bridged network adapter connecting the VM and the host machine.

closed 2016-07-08 by OwenArnold