ISISComputingGroup/lewis #56: Integration with IBEX

MikeHart85, 2016-06-16:
Depends on #53.

As a proof-of-concept, we want to implement a simulator of a simple device used in IBEX and configure it so that IBEX can interact with that simulation in the same way as it would if it was a real device.

This likely requires a TCP stream adapter.

MikeHart85, 2016-08-05:
This is largely done via #72.

To connect IBEX to a plankton simulated device, `st.cmd` in the IOCs `iocBoot` directory should be modified to remove or comment out this section:

```
drvAsynSerialPortConfigure("$(DEVICE)", "$(PORT)", 0, 0, 0, 0)
asynSetOption("$(DEVICE)", -1, "baud", "$(BAUD=9600)")
asynSetOption("$(DEVICE)", -1, "bits", "$(BITS=8)")
asynSetOption("$(DEVICE)", -1, "parity", "$(PARITY=none)")
asynSetOption("$(DEVICE)", -1, "stop", "$(STOP=1)")
asynOctetSetInputEos("$(DEVICE)", -1, "$(OEOS=\n)")
```

And replace it with this:

```
drvAsynIPPortConfigure("$(DEVICE)", "192.168.99.100:9999")
```

Where the IP and port is adjusted as required. If running on Windows, and using the Plankton Docker image, it should be launched with a `-p 9999:9999` (or whatever port number is being used) to forward the port from the container to the VM's network adapter.

MichaelWedel, 2016-08-11:
This has been achieved with the addition of the Linkam, so I'm closing this issue.

closed 2016-08-11 by MichaelWedel