ISISComputingGroup/lewis #81: Provide certain behavior as separate classes/functions

MichaelWedel, 2016-08-15:
Many default states for the devices have very similar code, because essentially they are "approach a target value linearly with a given rate of change". There should be exactly one place in the code where this is implemented.

My suggestion would be to create a new module in `core` for this. Maybe even in a sub-module called `behavior` or something along those lines.

MikeHart85, 2016-08-15:
We should consider non-linear approaches as well. Perhaps this could be implemented with a general `approach` function that takes a lambda/function reference to determine how exactly the approach is calculated. An `approach_linear` and similar could also be provided, which would call the general `approach` with appropriate pre-made functions.

Another thing to consider for such a library of general functionality is pseudo-random number generation. To keep the simulation deterministic even when "randomness" is required by device behaviour, we should have an easy way for the device developer to get pseudo-random numbers that are reproducible when necessary (and randomly seeded when not).

MichaelWedel, 2016-08-16:
Yes, let's think about how we could implement this in a good way. I like the idea of just exchanging the approach-function.

MichaelWedel, 2016-09-14:
This also needs to be part of Release 1.0 so that users have at least some functionality in this area.

PR #117 (2016-10-13): Linear approach function moved to core
This fixes #81.

There is now an additional `core.approaches` module, which contains a function `linear`. Others could be added easily. I tried implementing this as a class first with rate as a parameter to `__init__`, but the problem is that the rate is not always constant (see Linkam).
closed 2016-10-13 by MikeHart85