ISISComputingGroup/lewis #89: Update documentation once Communication Redesign completed

MikeHart85, 2016-09-08:
The front-facing `README.md` will need to be updated once design changes [discussed here](https://github.com/DMSC-Instrument-Data/documents/blob/master/design/plankton/CommunicationLayerDesign.md) are applied.

Perhaps this would also be a good time to split the documentation up a bit by topic, put most of the content in `/docs/`, and make the front-facing README leaner with links to the various topics. 

A concise overview with links to more tends to make for a more accessible entry point... Screenshots would also be a great addition, I think.

OwenArnold, 2016-10-11:
@MikeHart85 / @MichaelWedel I think this is somehow tied to the #108. I've added it in any case.

MichaelWedel, 2016-10-13:
Can this be closed now?

OwenArnold, 2016-10-14:
@MichaelWedel I'm waiting on @MikeHart85 to split the Framework Details off the front README. I'll do it if nothing gets pushed in the next couple of hours.

PR #123 (2016-10-14): Split README into separate sections
Closes #89.

The main README has gotten too long. This PR splits it into separate files that are linked from the main README, to make it a more accessible entry point.
closed 2016-10-14 by MichaelWedel