ISISComputingGroup/lewis #98: Ability to change setups at runtime

MikeHart85, 2016-09-20:
One aspect of #87 that was not addressed in #93 is switching setups at runtime.

Now that we have a way to control aspects of the simulation at runtime, it should be possible to load different setups of the same device at runtime as well. 

This would enable testing different failure scenarios without restarting the simulation. It could also be used to put a device into a mode where it just echoes the values you set to attributes, rather than actually running a real simulation (this could be done using a setup that forces it to stay in a neutral state that doesn't have any behaviour). This would also be useful for testing purposes.

This should be easier to do once creation of adapter and device instances is moved to the central `Simulation` class.

One technical challenge of this might be ensuring the adapter stays up, rather than shutting down and coming back online (if a new instance needs to be created).

MichaelWedel, 2016-09-20:
Just as a question, is it problematic to restart a simulation to test a different scenario? If the concern is that it would take too long or it would be too repetitive to get the simulation back to the state it was in, maybe addressing that could be another way.

I'm basically thinking of taking a snapshot of the device at a certain point in time (all the state, including current state of the statemachine) and applying that again, possibly to a device implementation with different state handlers in the state machine (for failure cases).

Regarding echo-mode, I somehow suspect that there might be other ways (tools) to achieve this. I'm not entirely sure how this "not simulating the device" fits with with our (current) goal of providing a device simulation. Maybe this is something for the long term.

MichaelWedel, 2016-09-20:
I had another thought regarding "echo-mode". Might this be solved at least partly be the logging improvements we've been discussing? Ideally, the Adapter would log at some level what commands it's receiving and the Device would in turn be logging what part of its interface is called. This could be inspected then.

MikeHart85, 2016-09-20:
The motivation is that the IBEX team would need to restart the IOC as well to restart the simulator, which would take too long when running dozens of unit tests.

> I'm basically thinking of taking a snapshot of the device at a certain point in time (all the state, including current state of the statemachine) and applying that again, possibly to a device implementation with different state handlers in the state machine (for failure cases).

That was my thought as well... it should be possible to do this at runtime, no? Create new instance (with different setup), copy attributes, replace references to new instances, old instance disappears.

Something we haven't really talked about during the redesign is how Setups would handle scenarios where I want to change how the interface behaves (so change what the T command returns for example). Do you have any thoughts on this?

I agree about echo mode not really fitting into our vision for Plankton but, as mentioned, it could be modelled just using a setup that never leaves the initial state or provides its own initial state without behaviour. So it doesn't really require any effort or changes on our part. It's just a convenient bonus that, if setups can be changed at runtime, the echo mode could be turned on and off at runtime as well.

Logging doesn't really fit with echo mode. The goal with it is to verify the IOCs ability to correctly encode and send data to the device, as well as successfully receive and decode data received from the device. With an echo mode, you can do this kind of round-trip test easily (set current temperature to 20.0, read current temperature, make sure it is 20.0). If the simulation is running properly, that value may well change between the time you set and read it. Then you don't know whether the value you read is legitimate or whether you made an error decoding it (from that weird ascii-hex-string format in the T status for instance).

OwenArnold, 2016-10-11:
@MikeHart85 @MichaelWedel Switching setups per say. Will not be part of release 1.0. Correct me if I'm wrong.

MichaelWedel, 2016-10-15:
Reading this again after a while I think we might need to clarify what a setup is. Currently a setup is a device class and constructor parameters. These in turn are currently options to override the state machine of a device, and data members.

I thiink switching that kind of setup at runtime should be relatively straightforward, we would need to generalize this "overriding capability" a bit, so that `Simulation` can apply the setup parameter to the device at runtime.

I'm not entirely sure about this, but this echo-mode sounds like entirely different behavior, almost like switching devices while keeping the adapter. This might work as well, but I probably wouldn't file this under setup anymore, at least not in the sense that we have setups at the moment.

Achieving this would involve a bit of work on the `Adapter`-classes, but maybe it's possible. The "binding step" between Interface and Device happens at runtime now anyway, all this would require is removing the forwarded members on the Adapter and attaching the new device. We have to think a bit about the simulation parameters though, would the cycle count/runtime be reset?

OwenArnold, 2016-10-19:
@MichaelWedel I suspect that the simulation must to be reset when the setup is changed. Anything else may be too complicated, or even impossible to handle correctly.

MichaelWedel, 2016-10-20:
I've given this some thought. In principle it would be possible to exchange a state handler that is not currently used without any larger impact on the simulation. The logic for this could be implemented in `StateMachineDevice`, so nobody would have to worry about this.

MichaelWedel, 2017-01-25:
Yesterday I tried out a few approaches for this. With the current way simulations are treated and how the lewis run script works, I don't think it's a good idea to implement setup switching.

To elaborate a bit. Given that "switching setups" basically means "switching device", it becomes automatically necessary to switch the adapter as well, because of the binding step. Furthermore we agreed earlier that the reasonable thing to do regarding simulation state would be to reset it (e.g. set cycles, runtime and uptime to 0).

Now if I haven't overlooked anything the easiest way to switch setups is actually to switch out the entire simulation. At that point the term "at runtime" also becomes questionable - runtime of what? lewis.py? The simulation? It wouldn't exist anymore. It must be the runtime of something else. Now we are back at the point where Lewis is essentially becoming a "simulation service".

I really think that this is an entirely different thing that requires some more thought. The lewis on/off service @MikeHart85 and I are working on is actually going in this direction and it could end up becoming some sort of "simulation service".

I'm moving this into the next release for now because I don't see a feasible way of implementing this at the moment.

OwenArnold, 2017-01-25:
I think I would be right in saying that it would be possible to do a complete tear-down lewis and a restart with the same ports just a different model for the device behaviour. At which point, I can see why you are  going in the direction of a service. I don't know much about IOC connection timeouts, but I guess timeouts could be an issue. I guess a service would also solve this problem. I think this needs more thought.

PR #212 (2017-03-16): Allow switching setups at runtime
This fixes #98.

With these changes it is possible to switch the setup at runtime via the control server. The functionality is implemented in `Simulation`.

I expect the implementation to change somewhat when all the adapter-related issues are taken up in the next release, so this implementation is very focused on providing only this new capability and nothing else.
closed 2017-03-16 by MikeHart85