ISISComputingGroup/lewis #105: Multi-disc choppers

dominikwerder, 2016-09-30:
I would like to propose some ideas for more flexible handling of multi-disc choppers:

Currently, the simulated chopper device exposes only the speed of the master chopper.  For multi-disc choppers, we would like the possibility to operate the discs individually phase-unlocked.

Also, some chopper electronics can keep the slave disc phase-locked at a certain speed ratio with respect to the master.

Furthermore, we would like the possibility to control more than two-disc systems, we have for example a five-disc chopper here.

One possible way to achieve this could be:

To duplicate the 'Spd', 'ActSpd', 'TDCE', etc for the number of available discs with a number suffix.

Phase-unlocking could be selected with Phs_[N] = -1.

A new field would be 'Ratio_[N]' (integer) to choose a possible speed ratio for the slaves.

What do you think?

Best regards
Dominik

MichaelWedel, 2016-10-03:
Hi Dominik,

thanks for your input! As far as we're informed each chopper disc will have its own CHIC, so syncronization would happen in a different layer. I will try and clarify with the Chopper group again.

To me it sounds like a relatively straightforward solution would be to create a different device that has the properties you describe. The device could create `N` instances of a simulated chopper internally and expose them via an EPICS-interface. The name could be `MultiDiscChopper` or something along those lines.

Changing the EPICS-interface of the simulated chopper as it is in plankton now is not easily possible because that interface is specified by ICS and the Chopper group.

We're in the middle of rewriting the way devices and adapters work (so not the best time to add more features :)), I suggest we talk about this during/after NOBUGS.

OwenArnold, 2017-01-30:
@dominikwerder is this still an active feature request?

MichaelWedel, 2017-02-15:
@dominikwerder, could I ask you to take a look at the way we implemented a chopper cascade in the `chopper_cascade` branch? It does not include a simulation, but the interface definition could be a step in the right direction.

dominikwerder, 2017-02-15:
Thank you, I'll look at it!

closed 2020-11-27 by mattclarke