ISISComputingGroup/lewis #135: Standardize Exception Handling

MikeHart85, 2016-10-26:
Currently, exception handling is very ad hoc throughout the system. 

The following exception classes exist:
* `StateMachineException` (statemachine.py)
* `RemoteException` (control_client.py)
* `ProtocolException` (control_client.py)

The latter two are probably fine as-is, since the control client is arguably its own application/library and those two exception types work well to capture and differentiate what can go wrong during its usage.

The simulation itself, however, needs better exception types and usage practices. Currently, only the StateMachineException exists as a distinct exception type. Any time an exception needs to be raised outside of that, standard classes like RuntimeError are used. *Even inside the StateMachine, sometimes a RuntimeError is raised instead of StateMachineException.*

This makes catching and handling exceptions difficult and, consequently, we don't do that very well either.

Ideas to consider/discuss:
* We should keep in mind that there are two distinct usage scenarios for the simulation: Command-line and as an imported library.
* Should we have a `core/exceptions.py` or similar to define all exceptions used throughout the system?
* Should we avoid allowing anything apart from exceptions we define to escape our code as a matter of principle? Raising standard exceptions within the system would be fine as long as they are handled before reaching the user under this model.
* One could argue that exceptions like AttributeError, ValueError, NotImplementedError, etc are descriptive enough and don't need to be wrapped in our own type / would be better as they are.
* Any of the above exceptions are meaningless if they come from too deep in our code. Getting an `IndexError` makes sense if you as the user provided a bad index. It doesn't make sense if you called a function that doesn't even take parameters, but something went wrong deep inside the core code (like, say, `pcaspy` couldn't be imported).
* RuntimeError, however, is so generic that I don't think it should ever escape our code (or perhaps even ever be raised?).
* On the other hand, we could consider not using our own exception types at all (including getting rid of StateMachineException) and just using standard exceptions for everything (in which case RuntimeError might be fine if there isn't anything more appropriate). This might make handling exceptions awkward though.
* When run from the command-line, we should have some high-level exception handling to make sure no exceptions escape to become traces. Errors should be displayed in a human readable way for the user, and ideally be specific enough to enable finding a solution easily.
* Interrupts like CTRL-C should also be handled gracefully.

This issue can be used to discuss how to best solve this.

This issue should probably be addressed after dealing with #125, #132 and #133.

MichaelWedel, 2016-11-02:
I agree with most of these points and while starting on #134 I already added a `core/exceptions.py` with one exception type `PlanktonException` (as suggested by you on slack). So my position on only standard exceptions vs. specific exceptions is very strongly specific exceptions for errors that are likely to be seen by users. It makes handling errors less awkward as you say and we'll be able to distinguish between high-level problems that we know about and we have a solution for and problems that are actual bugs.

That being said I think it's useful to allow exceptions to bubble up in general (i.e. not catching everything to shield it from the user). By presenting this in two different ways the user will know what to do through meaningful error messages when **they** did something wrong (malformed control server string, etc) and can open an issue or report otherwise in case **we** did something wrong, when they see a stack trace.

For many instances of the latter case the standard exceptions are fine, such as `ValueError` and so on.

You are right about `RuntimeError`, the places where we use that might be the first places to check for exceptions that need to be replaced with something more specific.

MichaelWedel, 2016-12-13:
The main action from this should be documentation both for users and developers. It should cover the exception rules for people in different roles:

 - Using plankton as an "end user", e.g. running `plankton.py` or `plankton-control.py`
 - Using plankton as a library
 - Developing devices
 - Core development

MichaelWedel, 2017-01-25:
I'm going through the remaining issues for 1.0.2 and I'm not sure what to do with this one - do we know well enough to add documentation at this stage? Stuff has happened since the ticket was created, we even have an `exceptions` module now.

If there are no concrete plans I think it's okay to put it in the backlog.