ISISComputingGroup/lewis #140: Documenting PCASPy installation

MikeHart85, 2016-10-28:
Since PCASPy is a bit quirky to install, we make sure our documentation covers how to install it.

This is particularly important since we will soon have a pypi package, where PCASPy is an optional requirement (or may be installed separately).

If I recall correctly, PCASPy requires `EPICS_BASE` and `EPICS_HOST_ARCH` to be set (as well as a corresponding EPICS-base installation of course). In the Docker build chain, these are set by `/etc/profile.d/01-epics-base.sh` provided by the `epics-base` image and sourced for pcaspy installation in the `plankton-depends` image.

@OwenArnold also mentioned some issues specific to OSX, where the installation was failing unless he compiled it from source.

MichaelWedel, 2016-11-02:
This can also be avoided by installing a slightly older version via `pip install pcaspy==0.6.1`.

MichaelWedel, 2016-11-02:
Regarding the documentation itself, maybe we should go back to the pcaspy docs and check whether we think they are sufficient and if yes, link to them. If not, we could add a very short section to our docs or we could launch a PR against their repo with improvements.

MichaelWedel, 2016-11-02:
This is their documentation:

https://pcaspy.readthedocs.io/en/latest/installation.html

To me this looks pretty much okay, it also explains the EPICS installation. Is it maybe enough to link to this?

MikeHart85, 2016-11-02:
Do you know what the issue on OSX is, exactly? We should probably raise an issue about it against their repository? We can recommend using 0.6.1 as a temporary measure, but it shouldn't be left like that.

Their documentation doesn't seem to cover `pip install pcaspy` on Linux at all. 

Instead it recommends using `easy_install` with a dated (2014) note about PyPi limitations (is it still true they PyPi doesn't support linux-specific wheels packages?).

But I agree it would be best to make improvements upstream and just link to their documentation rather than documenting ourselves.

MichaelWedel, 2016-11-02:
I think there's some hard-coded path somewhere that doesn't exist on my (or anybody else's) machine. You're right, we should report this. I will try to get the actual error message later today and write a ticket.

It is still true that PyPI does not support linux specific wheel packages unless they are pure Python. For packages which include any kind of build step, that needs to be done on the client machine. Somehow understandable given the fragmented world of Linux distributions. Nevertheless if EPICS and SWIG etc. are installed, running `pip install pcaspy` will perform all necessary build steps that are listed under "From source".

MikeHart85, 2016-11-02:
Yes, though you also have to set `EPICS_BASE` and `EPICS_HOST_ARCH`. And watch out that you use the `sudo -E pip install pcaspy` so that the environment carries over. Their documentation doesn't mention any of this.

Instead they recommend `easy_install` (which I think is deprecated?) and only mention the environment variables in the section for manually installing from source.

Or is their documentation correct, and we should use `easy_install` instead of `pip` for pcaspy?

In that case, our documentation is wrong, and our requirements.txt shouldn't contain pcaspy?

MichaelWedel, 2016-11-02:
I would in general recommend working in a virtual environment, then no `sudo` is required, but that's a bit beside the point.

I also seem to remember that `easy_install` is deprecated. I have never used it to install anything, at least I can't remember. Since installation via `pip` works (also on Linux, given that the build-from-source dependencies are installed) I think it's fair to assume that our documentation is not wrong :)

Would you mind opening an issue in their tracker to query them a bit? I hope they are actively using it: https://github.com/paulscherrerinstitute/pcaspy/issues

It's probably not that urgent, but it might be good to clarify. As written above, I will open a ticket for the Mac OS error later as well.

MikeHart85, 2016-11-02:
Sure, will do. I've opened an issue there before, and they responded quite quickly.

It looks like some people have also just opened pull requests and had them merged in the past. Might also be an option.

OwenArnold, 2016-11-25:
We discussed promoting the use of virtualenviroments in the install instructions.

MichaelWedel, 2016-12-22:
I'm moving this to the backlog, it doesn't seem so urgent at the moment.

closed 2020-11-27 by mattclarke