ISISComputingGroup/lewis #141: Make device Interfaces self-documenting

MichaelWedel, 2016-10-28:
Following some user feedback acquired by @OwenArnold that pointed out device `Interfaces` could be self-documenting with regard to the PVs/commands they expose.

This could be implemented in the respective Adapter, potentially using the docstrings of the exposed properties/methods. These can be extracted using the `inspect`-module.

dmichel76, 2016-10-28:
Yes, the idea I suggested to @OwenArnold was to somehow be able to query what the protocol is, as it's easy to forgot or not have the required documentation at hand. Happy to test it and give some feedback once implemented.

OwenArnold, 2016-10-28:
@dmichel76 I think it would be very useful to have you review this functionality. We'd definitely take you up on that.

MichaelWedel, 2016-10-28:
@dmichel76 thanks for the feedback, it's something we did not have on the screen so far. I like the idea of documenting these things because it also helps keeping the knowledge about a device alive somewhere.

Regarding the way this can be queried there are different possibilities. One is certainly the control server, but another one could be the `plankton.py`-script, which could in the first step simply print this information if a flag is provided, e.g.

```
$ ./plankton.py -p epics chopper --interface
Device: Chopper
Protocol: EPICS
PVs:
    - ...
    - ...
```

PR #148 (2016-11-14): Make devices self-documenting
This fixes #141.

This is my suggestion for "self-documenting" devices. `Adapter` now has an additional property `documentation` which can be overridden in sub-classes. For now I've decided against adding stuff to the class doc-string for several reasons, the most important being that the type of the interface is modified on construction and so it depends on the device and/or configuration (PV-prefix, host/port, ...).

Both `EpicsAdapter` and `StreamAdapter` auto-generate command lists from the `PV`/`Cmd`-objects. These now have `doc`-parameter that can be used to [... cut]
closed 2016-11-14 by MikeHart85