ISISComputingGroup/lewis #143: Setting multiple properties at the same time

MikeHart85, 2016-11-03:
A question that came up during a meeting with the IBEX team is whether multiple values could be set at once using `control.py` from the command line. 

My initial response was that this was a scenario where it might be more appropriate to write a script rather than using the command line utility.

Thinking about it, however, there may be scenarios where it is important to set multiple values atomically. That is, to ensure they are all set during the same cycle. 

Right now, due to the way the processing loop works, I think only one RPC command can be processed per cycle? If so, setting 10 different variables would get split across 10 cycles (1 second by default) even if you do it using a script. Even if this is not the case, it's impossible to ensure multiple properties are set during the same cycle.

We should consider:

* In the control client, making it possible to set multiple properties in one go using a dictionary or similar
* Exposing the above in the command line interface somehow (bonus, non-essential)
* Adjusting the way our processing loop works to permit multiple commands per cycle (the same goes for the adapters?)

With everything else going one, this is quite low priority for now, unless it becomes an issue in real usage, so putting it in the backlog.

MichaelWedel, 2016-11-03:
I agree that it is more of a backlog thing, but it should be easy. We could add a function in `Device`:

```
def set_parameters(self, new_parameters):
    for param, new_value in new_parameters.items():
        setattr(self, param, new_value)
```

Since we've switched to using `ast` for parsing, a dict can be passed via the command line:

```
./plankton-control.py device set_parameters "{a: 10, b: 20}"
```
Have not tested, might not work, but I suspect it might. Even without the method on `Device` similar methods could be added to each concrete device.

MichaelWedel, 2016-11-04:
After some syntax related experimentation I ended up with this which seems to work:

```
./plankton-control.py device set_parameters '{"name": "foo", "type": "bar"}'
```

MikeHart85, 2016-11-04:
Nice, I like it! :+1:

MichaelWedel, 2016-12-13:
This is something that could even go into `DeviceBase` now that this exists.

PR #182 (2017-01-25): Set multiple device parameters at once
This fixes #143.

A method `set_device_parameters` has been added to `Simulation` as per the discussion in #143. Documentation and release notes have been updated.
closed 2017-01-25 by MikeHart85