ISISComputingGroup/lewis #144: Adapter traffic being queued while disconnected

MikeHart85, 2016-11-03:
This is a bug reported by @IsabellaRey during an IBEX meeting.

Commands/requests sent to the device while it is disconnected are queued, then processed when it is reconnected. To reproduce:

1. Launch a `linkam_t95` simulator in plankton, with control server
1. Use control client to disconnect device
1. `telnet` into `linkam_t95` simulator, type `T` and press enter
1. Use control client to connect device

You will get a TArray reply after the last step.

This is a result of the fact that we don't actually disconnect but just stop processing the adapter. TCP/IP stack apparently still queues requests and gives them to us once we start processing again.

One possible solution is flushing all requests when reconnecting.

Side Issue: Step 3 arguably shouldn't be possible / doesn't make sense. Again, a consequence of not really disconnecting. Will make another issue for this.

MichaelWedel, 2016-11-03:
I did not expect this to work this way, I'm particularly unsure why step 3 works. We really do have to change this.

MikeHart85, 2016-11-04:
Yes, this is definitely unexpected and not really representative of any real-world scenario I can think of.

The reason step 3 works is because we simply stop calling `asyncore.loop()` when a device is "disconnected", but the `StreamServer(asyncore.dispatcher)` is still up (by design) and apparently it doesn't need loop cycles to receive `handle_accept` calls (I'm surprised by this).

Basically, I think there are two different scenarios we should simulate in this context:
1. Disconnect: The device is physically disconnected (or fails completely). In this case, all connected clients should be disconnected immediately and no further connections should be accepted. Attempts to connect should be rejected.
1. Suspend: The device stops responding/hangs/lags (potentially temporarily). In this case, current connections should stay open, but the device should stop responding. Attempts to connect should time out.

I'm not sure what should happen with commands sent in scenario 2 though (this issue, essentially). Should they be queued, as they are, or lost completely?

MichaelWedel, 2017-01-03:
The bug as described above has been fixed with #170, it is no longer possible to connect while the device is disconnected and connections are properly cut, so no queuing anymore. The suspend/resume issue is now in #171 - can we close this?

MikeHart85, 2017-01-04:
Well... really, we haven't fixed this so much as moved it from "disconnected" to "suspended" and temporarily removed the suspend feature. Should probably leave this open until #171 implements suspending in a way that fixes this. Or we could close this one, but then at least make sure #171 is clear that it's taking over this issue and is responsible for fixing it.

MichaelWedel, 2017-01-04:
Okay, I understand. In that case I will remove the highest priority label, because I don't think this is true anymore. This came originally from users wanting to disconnect the device, not suspend it. In the context of disconnecting, this is fixed.

MichaelWedel, 2017-02-07:
The originally reported bug has been fixed. I'm closing this because #171 covers the suspend/resume part.

closed 2017-02-07 by MichaelWedel