ISISComputingGroup/lewis #145: Could we implement real disconnecting as opposed to suspending?

MikeHart85, 2016-11-03:
Currently,  `disconnect_device` and `connect_device` functionality of the Simulation is implemented by suspending Adapter processing. The Adapter no longer gets cycles and so the device appears to stop responding.

Some issues with this are highlighted in #144.

Also, conceptually, this isn't really the same as disconnecting. 

The reason we took this approach is because of concerns about just closing and freeing up the port that the device simulator is listening on. If we simply free the port, other applications or simulators may claim it.

This issue is to explore two questions:

1. Would it be technically possible to better simulate a disconnect scenario without risking losing the port? For example, could all clients be disconnected and further connections refused?
1. The current behaviour could still be useful to simulate scenarios were a device stops responding (without being disconnected). But "disconnect" isn't really the right term for this. Should we rename current functionality to `suspend_device` / `revive_device` or similar?

MichaelWedel, 2016-11-04:
On first glance this appears somewhat adapter specific to me, because in EPICS this seems to behave like a disconnect - the PVs time out after a while.

What about introducing `disconnect`/`connect`-methods in `Adapter`? The default implementation could simply do nothing, so for EPICS we wouldn't have to do anything. For `StreamAdapter` we'd have to do something more drastic in there.

What are the options here? How much "risk" is it to give up the port for a short while? How would the dropping connections/refusing connections work technically?

MikeHart85, 2016-11-04:
I think introducing those methods on Adapter is the correct way to go. Maybe two pairs for slightly different behaviour:

* `disconnect` / `connect`
* `suspend` / `resume`

It's then up to the individual adapters how to implement those (and perhaps on some adapters one doesn't do anything or the two are synonymous).

The risk of losing the port can vary greatly depending on how you use it and what else is running on the system. In the context of IBEX, we talked about the central "simulator service" a while back, from which you could request a simulator for a device, and which would then find an open port (presumably starting at a specified port like 20000). In this case, there's a high chance your port could be hijacked.

I don't know about `asyncore`/`asynchat` specifically, that would be something to research (or perhaps even use a different library for the networking if necessary). But in general, closing an open socket connection should be trivial. Refusing connections on an open port might be trickier... we may need to accept the connection and then immediately close it as an approximation.

PR #170 (2017-01-03): Fix disconnecting devices
This fixes #145.

`Adapter` now has a method `stop_server` in addition to `start_server`, as well as a `is_running` property which indicates the status of the adapter. Both in the EPICS and Stream cases, the entire server object is completely torn down in `stop_server`, so that existing connections are closed and new connections can not be made. As discussed in #144 and/or #145, there is a risk that the port is blocked by another process in the meantime, but currently all of that is the user's responsibility anyway so at the moment I do not see a big problem with this [... cut]
closed 2017-01-03 by MikeHart85