ISISComputingGroup/lewis #151: Error displayed when Device not found

MikeHart85, 2016-11-11:
If you try to start a device that doesn't exist or cannot be found, e.g.:

```
$ ./plankton.py foobar
```

You will get an error like this:

```
An error occurred:
Could not find setup 'default' for device 'foobar'.
Are you sure the correct device package is used (flag -k) and that it is in the path (flag -a)?
```

This can be confusing, especially when the device you're trying to load has no setup. It should really say that it couldn't find the device, and not mention setups.

MichaelWedel, 2016-11-11:
I'm tempted to say we can fix it as part of #132, where all of these things are going to change anyway.

MichaelWedel, 2016-12-13:
This was resolved in PR #154, the error is now:
```$ ./plankton.py foobar
An error occurred:
No device with the name 'foobar' could be found. Possible names are:
    chopper
    linkam_t95
See also the -k option to add inspect a different module.```

closed 2016-12-13 by MichaelWedel