ISISComputingGroup/lewis #160: Create a MODBUS interface

DominicOram, 2016-12-14:
At ISIS we are trying to write an emulator for a Eurotherm nanodac http://www.eurotherm.co.uk/products/recorders/graphic/nanodac. We plan to talk to the device using MODBUS TCP and already have an IOC (written elsewhere). The use case for writing an emulator is that we have yet to purchase the equipment and we would like to test the IOC.

We are happy to do the actual device emulator ourselves but it would require MODBUS support within plankton.

MikeHart85, 2016-12-14:
Is the IOC available somewhere, for reference and testing?

DominicOram, 2016-12-14:
The support module is at https://github.com/ISISComputingGroup/EPICS-nanodac and an example IOC at https://github.com/ISISComputingGroup/EPICS-ioc/tree/Ticket1843_nanodac_support/NANODAC.

PR #175 (2017-01-25): Support for Modbus Protocol
Fixes #160.

This PR adds a Modbus Adapter and an example device that uses it.

Several Modbus libraries were considered, but all of them had issues that disqualified them. `pymodbus` was the most feature complete Modbus implementation and was very attractive. Unfortunately, it doesn't support Python 3 as of this writing. Consequently, we decided to implement a small subset of Modbus ourselves and expand it as required.

References used:
* http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf
* http://www.modbus.org/docs/Modbus_Messaging_Implementation_G [... cut]
closed 2017-01-25 by MichaelWedel