ISISComputingGroup/lewis #190: Multiple interface, interface documentation

OwenArnold, 2017-01-30:
Follows on somewhat from #141, for devices with multiple interfaces it's not clear how you continue with the `-i` option to get the information from one of the interfaces. For example:
```
(venv)/Users/spu92482/sources/lewis (master)$./lewis.py --show-interface julabo
2017-01-30 19:10:54,612 INFO lewis.DeviceBase: Creating device, setting up state machine
An error occurred:
Failed to find protocol 'None' for device 'julabo'. Available protocols are: 
    julabo-version-2
    julabo-version-1
```

Needs to be discussed before being assigned a milestone and priority.

MichaelWedel, 2017-01-31:
It should work using `lewis -p julabo-version-2 -i julabo`.

MikeHart85, 2017-01-31:
Perhaps just a matter of rewording the error slightly to something like:
> Must specify protocol (using -p option) for devices that have more than one. Available protocols are: ...

OwenArnold, 2017-01-31:
@MikeHart85 yes, that was what was missing.

PR #193 (2017-02-10): Clarify error message for multiple interfaces
This fixes #190.

The error message has been reworded, including a hint to the -p option as suggested in the ticket.
commit 91cf86cc (2017-02-10): Clarify error message for multiple interfaces

This fixes #190.

The error message has been reworded, including a hint to the -p option as suggested in the ticket.
commit 8a844dbf (2017-02-15): Clarify error message for multiple interfaces

This fixes #190.

The error message has been reworded, including a hint to the -p option as suggested in the ticket.
closed 2017-02-10 by OwenArnold