ISISComputingGroup/lewis #194: More selective remote exposure of attributes/methods

MichaelWedel, 2017-02-08:
#191 addressed a few issues that became apparent after we started using `ControlClient` in other contexts. `ControlServer` makes very few assumptions on which parts to expose and which ones not, which I think is okay because that way it stays quite simple and general. As pointed out by @MikeHart85 via Slack, this does lead to some annoyances when using `lewis-control`, particularly with non JSON-serializable members.

Since we're using JSON-RPC as a transport it's probably fair to assume that only JSON-serializable members should be exposed - all others are as of now unusable anyway. This should already take care of quite a few "annoying" cases (such as the `log` member which was pointed out specifically). This partially reverses one aspect of #191, each attribute would have to be inspected once on the server side to determine whether it should be exposed or not. I think that is acceptable, the majority of property getters should not have side effects.

When it comes to methods, it gets a bit more tricky, because there it's more of a semantic question of what is allowed and what isn't. We're already not exposing members that start with `_`, because they are considered private and that makes sense. But the solution can't be to make everything private just so it's not exposed via `ControlServer` (that might not even be active).

`ExposedObject` already has the `expose` and `exclude` parameters, which give some control over the parts of an object that should be exposed and `Simulation` already uses this. I think this is the right point to start looking at - making it easier to `exclude` stuff. Maybe `exclude` could even accept types and exclude the entire interface of the type - that would work for `CanProcess` and/or `CanProcessComposite`.

MikeHart85, 2017-02-09:
Could we not just "subtract" all members provided by / shared with `Device`, `has_log`, etc to only expose the ones provided by the device implementation itself?

>Since we're using JSON-RPC as a transport it's probably fair to assume that only JSON-serializable members should be exposed - all others are as of now unusable anyway.

I think it would be better to still show the unserializable ones that are provided by the device implementation (with the error value, as it currently is). A user might not be aware unserializable members aren't exposed, or simply not realize a member happens to be unserializable (`set`, for example, was very unexpected to me). In this case, it would be very confusing if we just silently hide those members.

MichaelWedel, 2017-02-09:
> Could we not just "subtract" all members provided by / shared with Device, has_log, etc to only expose the ones provided by the device implementation itself?

Yes, that is what I suggested in the last paragraph, `exclude` could also accept types in the list. Then we could just exclude everything from `DeviceBase` and `CanProcess` (and whatever else we see fit).

PR #213 (2017-03-16): Filter Members Exposed via Control Interface
Fixes #194.

Filters members exposed by default (not overridden via the `members` param) via `ExposedObject`.

Members filtered, in addition to the previous "starting with `_`" condition, are anything defined or inherited in `Device` or `StateMachineDevice` and specifically "`log`". 

`log` had to be done explicitly since it is added dynamically by a decorator.

Also fixed an error in `control.py`... not sure how that slipped past both of us! :laughing:
closed 2017-03-16 by MichaelWedel