ISISComputingGroup/lewis #204: Maintain devices in separate package

MichaelWedel, 2017-02-21:
We've discussed this earlier and after thinking about it for a while I believe it's a good idea to separate the devices into their own package that can be installed and maintained.

This consequently drives Lewis more in the direction of a library rather than a simulator for a few chosen devices.

There are some things that need more thought beforehand, for example unit tests. I think the unit tests for devices should be part of the device package like so:

```
devices/
  device_1/
    interfaces/
    other_stuff/
    tests/
  device_2/
    interfaces/
    tests/
```

Running these tests could be part of the "device verification" suggested in #199.

MichaelWedel, 2017-02-21:
I did not want to put it into 1.0.3 just yet, there a quite a few unanswered questions or other issues that need some work first. So it's in the backlog for now with the option of moving it to the release should it be realistic.

MikeHart85, 2017-02-21:
Perhaps it would make sense to restructure devices so the top folder is empty (or perhaps just contains a config file or the like):
```
/
  device_1/
    device/
    interfaces/
    setups/
    tests/
  device_2/
    device/
    interfaces/
    setups/
    tests/
```

MichaelWedel, 2017-02-21:
We just have to keep the packaging in mind and it also depends a bit how much "stuff" there is at the top level. If there are docker files, travis config and so on I'd probably prefer doing it like it is with Lewis now (with a directory).

We'll see how it develops.

MikeHart85, 2017-02-21:
No, I meant the top level inside each `device_1`, `device_2`, etc. Not the top top level. Sorry, confusing wording.

MichaelWedel, 2017-02-21:
Ah okay, now I see. An empty `__init__.py` file might still be required so that the import stuff works properly.

closed 2020-11-27 by mattclarke