ISISComputingGroup/lewis #226: Documentation for Setups

OwenArnold, 2017-03-27:
With the versioning introduced in `release1.0.3`, we anticipate devices/interfaces living in a separate package. To make this even more generic, we want people using setups to add any edge-case behaviour over the vanilla devices. Devices should be written to be as close as possible to the real thing. We therefore need to improve our documentation for setups. I've discussed with @MichaelWedel putting something into the developer documentation to fully explain what you can do with setups with some nice examples. i.e how to add your own custom `State`s and `Transition`s

MikeHart85, 2017-03-28:
This is a good idea, we currently neglect setups a bit.

It might also be a good opportunity to take another look at whether setups are capable of doing everything that device developers expect / need from them, or whether we need to revisit how they work and what they can do.

MichaelWedel, 2017-03-28:
Yes, we should definitely update the documentation. Whether setups do everything they need to do is a separate (and potentially more difficult) question that probably should get its own issue.

closed 2020-11-27 by mattclarke