ISISComputingGroup/lewis #231: Create a Serial Stream Adapter

DominicOram, 2017-04-18:
At ISIS we have legacy drivers that talk to devices that are connected via a serial port. For testing purposes it would be useful to write an emulator that these drivers can communicate with. We are currently doing this by running another script to convert Serial <-> TCP, see diagram below

[image]

Note that the physical loopback is unfortunately required for the OS to properly recognise the port. 

It would be useful if Lewis could communicate using serial, therefore removing the need for the script. I am happy to do the work for this, if you think it's something that would benefit Lewis.

MichaelWedel, 2017-04-19:
Thanks for taking the time to write up the issue, I think it's an interesting use case and would enable Lewis to be used for testing a larger variety of higher level software.

Is the TCP <-> Serial script available somewhere? It would be interesting to take a look and maybe it already provides a base for an implementation.

I'm adding it to the current milestone and some tags as well.

MikeHart85, 2017-04-19:
The script is available here: [com2tcp.py](https://github.com/ISISComputingGroup/EPICS-DeviceEmulator/blob/47486e04f2d596f58b2c0882d9243d84e1fcfd4d/com2tcp/com2tcp.py)

The related IBEX issue is here: [Emulator: Connect LabView drivers to emulator](https://github.com/ISISComputingGroup/IBEX/issues/1899)

FreddieAkeroyd, 2018-01-09:
Is it OK for me to take a look at this?

MikeHart85, 2018-01-09:
Absolutely!

I think we left this be mostly because the com2tcp script approach seemed to be working and actually looked like an elegant approach, as far as amount of code required to accomplish the goal goes.

An integrated solution would be cleaner in terms of less interacting components, but will likely require much more code / refactoring of existing code.

You'll want almost all of the code from the TCP Stream adapter, except have it connect via serial port somehow. Last time I looked, I remember not seeing a clean way to do this, without either copy-pasting huge chunks of stream adapter code into a new serial adapter, or some major refactoring to allow two adapters to share the same code.