ISISComputingGroup/lewis #276: Any new improvements?

chansdad, 2019-09-03:
This is one of the many useful projects i have seen in the age of IOT. anyone working on this actively? Would like to have some new features added to it 

Thanks

mattclarke, 2019-09-04:
Hi.
We are not doing much with lewis at the moment because it currently meets our requirements. 
However, some of your suggestions are interesting, so we will look into them next time we are working on the code base.

Thanks for getting in touch :)

chansdad, 2019-09-05:
Thanks . Repurposed it for the time being , by writing the device objects into a array .. While in my case devices dont interact with one another , would be good to really utilize the state machine to set different conditions in the device . For now each of the objects state is updated by way of updating its properties. Further more one of the key things i noticed is state is not persisted across restarts which is significant ..  Also telnet is a big no no to control device behavior .I warpped the commands in a rest api . Kind of hard coded for now , I am not an expert level python programmer , hacked my way and got it to do what i wanted , but this product sure has a TON of potential with the right set of features..

mattclarke, 2019-09-06:
Can you create a draft PR, so we can see what you have done please?
It might be something we could use/add

chansdad, 2020-02-21:
This was on my plate for a while , I will start cleaning up a bit and will write up something .

closed 2020-11-27 by mattclarke

chansdad, 2020-12-15:
Unfortunately did not get time to create a draft PR . are you still enhancing this project?

mattclarke, 2020-12-15:
Not so much, it currently has the functionality the main contributors generally need.
But if we require new functionality then we will implement it.

Still happy to look at and discuss PRs from other people though :D