ISISComputingGroup/lewis #281: Binding devices attributes via modbus

KacperKlys, 2019-10-16:
Hi guys, 

In the documentation file it is written that: 

> class `ModbusBasicDataBank` does not support binding addresses to attributes or functions of the device or interface.

So my question is what does support binding mechanism in case of modbus.
In the example on github, there is only interface layer with `ModbusBasicDataBank` which initializes addresses and values.
How I can get data from simulated device via modbus?

DominicOram, 2019-10-16:
Hello,

We have an example of how to create a modbus device [here](https://github.com/ISISComputingGroup/EPICS-DeviceEmulator/tree/master/lewis_emulators/moxa12xx). The `interface.py` is setting up the Modbus interface with which registers to use and `device.py` will populate those registers. I hope that example helps, if it doesn't please let me know and I'll write a more basic example.

KacperKlys, 2019-10-21:
Hi Dominic, 
Basing on the link you sent me I tried to do something on my own. Unfortunately, I think i'm still missing something since I can't get rid off one error. 
My short project is made of one file `chiller.py`. Problem is in line 43 in `chiller.py` - 

> `Chillerdevice` has no attribute `interface`

Here is my [code](https://github.com/KacperKlys/ChillerDriver/blob/master/chiller.py)

DominicOram, 2019-10-21:
Hello @KacperKlys. I've taken your example and got it to work in https://github.com/DominicOram/LewisExample.

The main issue was that you need to [manually assign the interface to the device so that it has a reference](https://github.com/DominicOram/LewisExample/blob/f53f8fcec121a80525421b3bbbd197ff1419f2d6/interfaces/interface.py#L7). This may make sense as something the framework does for you, I'll have a think about adding it.

I also fixed a couple of minor typos in your device, it should all now work fine. Please let me know how you get on. Out of curiosity what is the device that you are emulating? We should probably come up with a shared set of common devices.

KacperKlys, 2019-10-27:
Hi @DominicOram ,

I have created first version of my device's simulator [ChillerSimulator](https://github.com/KacperKlys/ChillerSimulator). I would be grateful for any comments or tips. 

I still have some questions. I thought that the idea of this framework is to have seperated layer of device and layer of ccommunication so we can create different types of interfaces for the particular device without changes in the code of device. In this case, the communication layer is associated with the device. Is there any way to rewrite this code to seperate those layers?

Answering your question, this is chiller for control system of lasers, this device will be connected via modbus and then controlled by EPICS.