ISISComputingGroup/lewis #287: Create an adapter which communicates with API's

dheeraj539, 2020-09-21:
We have many devices that communicate using the vendor specific API's.
[image]
Is there a best way to do simulation of this type of devices using LEWIS?

mattclarke, 2020-09-21:
Hi and welcome,

Interesting question, I think we need a bit more information:

- Are you linking against the vendor's libraries in your code?
- Do you know how the vendor libraries communication with the device? For example, via ethernet, COM etc?

Cheers,

Matt

dheeraj539, 2020-09-21:
Are you linking against the vendor's libraries in your code?
_**Yes, I am linking the vendor libraries in my code.**_
Do you know how the vendor libraries communication with the device? For example, via ethernet, COM etc?
_**They communicate using the USB serial ports.**_

mattclarke, 2020-09-23:
I assume the vendor library is closed source?

One way might be to wrap the vendor's API so that it could be communicated with over a network using a protocol like EPICS (https://epics.anl.gov/) and have your code call that.
That should be possible with LeWIS as it supports EPICS; however, it would require you to wrap the vendor's library with pybind11 or to use ctypes. Then LeWIS could be swapped between talking to the real device and a simulator.

This might be a lot of work though as you would have to also make changes to your "client" code to communicate with EPICS.

dheeraj539, 2020-09-25:
I assume the vendor library is closed source? - _**Yes, it is.**_

Thank you for the suggestion. Could you please elaborate bit more(may be some sort of flowchart)? I also didn't get "vendor API(which works on USB interface) to communicate over network".  The said device is a spectrometer. Thanks in advance

mattclarke, 2020-09-25:
[image]

Basically what you have now is a thick client, i.e. one program that has all the code in. My suggestion is to make it a client server application where the server consists of the Vendor API plus a wrapper that allows communication via TCP/IP. The client would have the rest of your code in it and would send commands and receive information from the server via TCP/IP.

This could be done just using plain network sockets or by making the server a webserver, or using EPICS.

mattclarke, 2020-09-25:
LeWIS could be the server if you use EPICS and the wrapper could be written using pybind11 or ctypes.
The client can remain as C++, but you would have to link against the EPICS libraries.

closed 2020-11-03 by mattclarke