ISISComputingGroup/lewis #320: remove asynchat/asyncore and use asyncio instead

rerpha, 2024-11-13:
If we want to use python >=3.12 we need to remove this as it's been deprecated until then and completely removed in 3.12 onwards. 

it's currently used by the stream handler, #306 says a bit about this and how we might change it to suit. 

### acceptance criteria
- [ ] asynchat and asyncore are removed from use in favour of asyncio
- [ ] our python testing matrix tests against 3.12 and beyond
- [ ] the pyproject.toml is updated to use `>=3.10` as its python-versions

Tom-Willemsen, 2024-11-15:
Short term fix for 3.12 could be to use a shim (https://pypi.org/project/pyasynchat/), but should still do the refactor towards standard asyncio longer term

anderslindho, 2026-02-24:
Is there any plan to work on this at ISIS? Asking because we otherwise might want to prioritise doing it at ESS - we have a want for this conversion to remove complicated dependency management.

closed 2026-08-25 by Tom-Willemsen