ISISComputingGroup/lewis #322: Simulation stop raises RuntimeError "Cannot join current thread".

pheest, 2025-01-02:
The simultaion stop() method is called within the context of the control server thread.
It calls _stop_control_server, which calls _control_server_thread.join().

This means that the control server thread is seeking to join to itself.
This is not permitted, see https://docs.python.org/3/library/threading.html .

Steps to reproduce:
1. Launch a simulator instance from a command window with e.g. set C:\Python311\scripts\lewis.exe linkam_t95 -r localhost:10000
2. From a second command window 'C:\Python311\scripts\lewis-control simulation stop'.
The simulation exits with exception raised. This is logged by both command windows.

I was able to address the issue locally by moving the call to _stop_control_server out of the 'stop' method and into the 'start' method - after the while not self._stop_commanded: loop.
Which is within the simulation main thread, rather than the control server thread.

This allows the simulator to stop in a clean and error-free manner.

As a workaround, it is possible to catch the RuntimeError exception.

closed 2025-03-18 by rerpha

Tom-Willemsen, 2025-03-18:
Hi @pheest - sorry for taking so long to notice this issue. We've just published `1.3.5` to pypi which contains a fix for this.

pheest, 2025-03-19:
Thank you, Tom. Much appreciated.